tracespace/tracespace #364: Unexpected layer detection

M0hammedImran, 2021-05-19:
I have a 4 layer board Gerber file that I used to plot SVG but all the files except drill file is detected as copper layers which results in the output saying I have 10 layer board.
[STM_2020_30_12.zip](https://github.com/tracespace/tracespace/files/6509991/STM_2020_30_12.zip)

kasbah, 2022-02-06:
Seems like a dupe of https://github.com/tracespace/tracespace/issues/357, will add your example there.

kasbah, 2022-02-06:
Actually, your problem is also that these names are not detected. Can you explain where the convention used in the zip comes from @M0hammedImran?

M0hammedImran, 2022-02-07:
This was given to me by one of my client. I think they used [Autodesk Eagle](https://www.autodesk.com/products/eagle/overview).
It's a relatively popular tool.