tracespace/tracespace #365: Incorrect parsing of Outline

dottenbr, 2021-06-09:
### This is the outline layer I have

[image]

### This is what tracespace turns it into

[image]

### This is the file

[buggy.GM1.zip](https://github.com/tracespace/tracespace/files/6623249/buggy.GM1.zip)

mcous, 2021-06-10:
I can get the board to render correctly if I set the gap fill setting to `3.6 mm`. For looking at the outline layer and making the line thickness smaller in the dev tools, there looks to be some fairly sizable gaps between the endpoints of the corner arcs and the line segments.

tracespace's board outline parsing is really naive; it only really works if all the individual segments of the outline layer line up. It cannot account for gaps this large with its default settings.

dottenbr, 2021-06-10:
Thank you for this super quick response. I wonder, does such a high gap fill setting cause other unintended effects? Maybe the default could be increased then.

This is a serious issue with the Gerber format in general that it can be sometimes a bid janky. 

As a team we will think if we can come up with a good proposal of how this could be improved. 

At least I know what to do when this happens.

mcous, 2023-01-22:
Testing with the new board shape plotter in `@tracespace/core@next`. With the (completely arbitrary) default settings, this board still does not render properly.

Gaps:
- gap of `0.03999999999999915` between `26.989,23.5` and `26.949,23.5`
- gap of `0.09999999999999787` between `26.989,0` and `27.089,0`
- gap of `0.13899999999999935` between `16.951,37.745` and `17.09,37.745`
- gap of `7.369883920388437` between `26.949,23.5` and `21.519,18.517`
- gap of `0.05800000000000005` between `1.042,0` and `1.1,0`
- gap of `0.08584287972802494` between `16.078,36.83` and `16.09,36.745`
- gap of `1.4142135623730951` between `16.09,36.745` and `17.09,37.745`
- gap of `5.9559999999999995` between `27.089,0` and `21.133,0`
- gap of `20.032999999999998` between `1.1,0` and `21.133,0`

<details>
<summary>Full outline plot</summary>

```json
{
  "regions": [
    {
      "type": "imageRegion",
      "segments": [
        {
          "type": "arc",
          "start": [
            21.519,
            18.517,
            0
          ],
          "end": [
            21.519,
            18.517,
            6.283185307179586
          ],
          "center": [
            20.573999999999998,
            18.517
          ],
          "radius": 0.945
        }
      ]
    }
  ],
  "openPaths": [
    {
      "type": "imagePath",
      "width": 0,
      "segments": [
        {
          "type": "arc",
          "start": [
            28.008,
            22.481,
            0
          ],
          "end": [
            26.989,
            23.5,
            1.5707963267948966
          ],
          "center": [
            26.989,
            22.481
          ],
          "radius": 1.019
        }
      ]
    },
    {
      "type": "imagePath",
[... 265 lines cut ...]
        },
        {
          "type": "line",
          "start": [
            10.236,
            0
          ],
          "end": [
            1.1,
            0
          ]
        }
      ]
    }
  ],
  "size": [
    19.628999999999998,
    17.572,
    21.519,
    19.462
  ]
}
```

</details>