tracespace/tracespace #367: Can't seem to get it to work on my gerbers.

bryancostanich, 2021-07-14:
Hey folks, apologies in advance if I'm totally messing up the usage of this, but I'm trying out the **gerber-to-svg** stuff to turn a gerber from EasyEDA into something that I can laser on my glowforge and am really struggling.

I have these [AISU_Gerbers.zip](https://drive.google.com/file/d/1EwzHFkdw-secxbcfP1qyynzud_5Plxra/view?usp=sharing) and i'm specifically trying to convert the `Gerber_TopPasteMaskLayer.GTP` to an SVG. I've used both the tracespace.io site (which is very nice, by the way), and gerber-to-svg from the command line and I'm getting the same output. When I import into the glowforge app, i'm seeing incomplete shapes:

[image]

When I opened up in figma, they appear like maybe they're correct.

[image]

Except that they're all filled. And the entire SVG is _very small_, `2.38` x `2.02` pixels, in fact.

So after making it much bigger so i could examine it, I'm noticing some other funky things. The first is that the stroke width for all the shapes have a thickness of `0`:

[image]

Removing the fill and changing the stroke width to 1, i see that some of them are in fact, incomplete outlines:

[image]

Any idea of what i'm doing wrong here?

kasbah, 2021-07-15:
Not sure what's going on with gerber-to-svg there but just in the interest of getting you the SVGs you want. Have you tried [gerbv](http://gerbv.geda-project.org/) and printing to SVG?

kasbah, 2021-07-15:
I printed the paste and outline layer with gerbv in case it saves you some time: [gerbv-print.zip](https://github.com/tracespace/tracespace/files/6824031/gerbv-print.zip)