tracespace/tracespace #370: How to handle .GML milling layer from Eagle?

kasbah, 2021-07-15:
- https://github.com/strawlab/Miriam/tree/main/PCB/Shield
- [on tracespace.io](https://tracespace.io/view/?boardUrl=https://rawcdn.githack.com/strawlab/Miriam/582432f1fe42adff0942121773e72ed82a354bb0/PCB/Shield/Gerber/shield_008.zip)

The files have a .MIL layer to drill some slots. It's recognized as an outline but there's a separate .GM1 for the outline. I guess this is a case for merging these layers?

mcous, 2022-06-29:
Oh hm, I think tracespace should be smarter and only select one outline file out of potential candidates. I think to do so reliably, we're going to need to build a much better outline analysis pipeline

shawon1220, 2022-07-13:
I am now looking for the outline layer with the largest width

shawon1220, 2022-07-27:
> hm, I think tracespace should be smarter and only select one outline

If  can find the coordinates of the effective contour, it is very useful to calculate the border size and preview the pcb board picture. My current idea is to use opencv to find the most effective contour. I can't think of how other platforms can accurately calculate the coordinates Except for board size.