tracespace/tracespace #371: fails to load gerbers from Allegro 17.2-060 (Gerber RS274X*)

rgetz, 2021-07-26:
I have a few gerbers from Allegro, which fail to render/load.

Entire fab package:
[cn0536-designsupport.zip](https://github.com/tracespace/tracespace/files/6881161/cn0536-designsupport.zip)

specific files:
  * `CN0536-DesignSupport/fab/1_primary.art`
  * `CN0536-DesignSupport/fab/l2_gnd.art`
  * `CN0536-DesignSupport/fab/l3_vcc.art`
  * `CN0536-DesignSupport/fab/l4_secondary.art`
  * `CN0536-DesignSupport/fab/silk_prm.art`
  * `CN0536-DesignSupport/fab/silk_sec.art`
  * `CN0536-DesignSupport/fab/ncdrill-1-4.drl`

those were able to be loaded with gerbv (except for the drill files).

Thanks

jaseg, 2022-06-21:
@rgetz I have included your files in the gerbonara testsuite at https://gitlab.com/gerbolyze/gerbonara/-/commit/0f70b225cc394338e9c37c1b1c34c3e73609f7d6 . I hope that's ok with you, otherwise please ping me and I'll remove them.

gerbonara is a python gerber library unrelated to tracespace, but I found this issue while working on a comprehensive test suite.

rgetz, 2022-06-21:
Thanks for feedback. 

It’s totally fine to use these for any purposes - I get that Allegro isn’t exactly the most popular tool in the community - but it is the one that most semiconductor manufactures use - so many eval boards and other semiconductor reference designs are in that format. 

- Robin

jaseg, 2022-06-27:
@mcous  FWIW, I am not a fan of Allegro's gerber export style either, but it can be parsed unambiguously. The tool definition comments in the excellon file are always the same format, and the excellon file's number format can be read from either the accompanying nc_params.txt or ncdrill.log file. [Here](https://github.com/jaseg/gerbonara/blob/main/gerbonara/excellon.py#L579) is how gerbonara parses Allegro tool definition comments and [here](https://github.com/jaseg/gerbonara/blob/main/gerbonara/excellon.py#L87) is how gerbonara parses Allegro's ncparam.txt and ncdrill.log files (next function below).

I think the greater issue with Allegro is that it does not seem to ship with a pre-defined gerber file naming scheme, so everyone just makes up their own and automatically matching allegro gerbers to layers is mostly guesswork.

mcous, 2022-07-25:
Drill file support for allegro now tracked in #392