tracespace/tracespace #374: Recognising non-copper layer as copper layer

Upendrasengar, 2022-02-06:
It's is happening for some files where it is provided with non-copper layer files but in JSON we see the type is a copper layer.  Not sure if it is earlier mentioned or not but this has been a long-running issue for me.

Any insight would be helpful.

kasbah, 2022-02-06:
Seems like the same issue as https://github.com/tracespace/tracespace/issues/357, if you have specific examples please add them there.

closed 2022-02-06 by kasbah

Upendrasengar, 2022-02-21:
@kasbah thanks for the response.
can you please help me identify the issue, I'm attaching the file if you can please verify
[test-file.zip](https://github.com/tracespace/tracespace/files/8107202/test-file.zip)
?

Upendrasengar, 2022-02-23:
@kasbah can you please update anything here?