tracespace/tracespace #375: Tracespace displays unknown copper element (duplicated pin) on a rotated footprint from gerber

netquik, 2022-03-05:
Don't know if it could be a bug/issue on kicad or tracespace.

I had a problem with tracespace displaying and exporting an "unknown" copper element that belongs to a footprint.
Gerber has been exported from kicad. No clue about that copper on pcbnew or kicad gerber viewer.
this is tracespace on front gerber
[image]
this is kicad gerber viewer  on the same file
[image]

it seems (must be) a pin of the footprint that is added (not rotated)  in that position across others 2 pins.
I tried deleting those 2 pins to see. The ghost pin appears crossing previous 2 pins. 

After many tries **i managed to have it displayed correctly** by setting 0° rotation on kicad and 90° again and re-exporting gerber.
Strange thing was that kicad gerber viewer displayed that file correctly even before.

Thanks.
I'm reporting it only for people that may experience the same problem.

kasbah, 2022-03-05:
Thanks for reporting, could you check with gerbv to compare (or share the relevant file)?

netquik, 2022-03-05:
I will try. I have some backups so i should recover the problematic file. I will update as soon as possible.

netquik, 2022-03-05:
Well not easy to reproduce. I found the bugged file on old version of the board. Then i tried to solve the problem as i did before but it did not work. After some changing in rotation of the two footprint IC1 and IC2 I came with two file that should be identical in kicad but display different on tracespace. Note that both files have the issue on IC2 while only one have issue on IC1. I'm attaching two files.
[GERBER.zip](https://github.com/tracespace/tracespace/files/8191149/GERBER.zip)

i can't use gerbv but i compared the files normally. There are some difference regarding SMD PADs that have issues.
opening them with kicad gerber viewer does not show the problem

netquik, 2022-03-05:
Also note that the issue is back again on tracespace on my last board version (on both rotated IC). 
If i set rotation to 0 the problem disappears. I'm lost in solving this.

netquik, 2022-03-06:
So i did many tests. No way to understand the problem.
Anyway the issue shows only on gerber PROTEL files.
**GBRs does not have this problem of not rotated duplicated smd pads**
I didn't realize that because tracespace seems to work on copper layers with custom names.
I had to revert to F.Cu and B.Cu.

netquik, 2022-03-10:
Updating you on the issue.
Well the problem came up again  from nothing (I did not touch the interested footprints) and also in .gbr files.
No other gerber viewer shows the problem. (even ucamco).

[image]

kasbah, 2022-03-10:
Can confirm that, with the gebers from the zip attached above, I can see the problem on tracespace.io but not in gerbv.

netquik, 2022-03-10:
Maybe i found the issue
Tracespace fails in some way with the NET name PD3

so

```
%TO.P,IC2,9,PD3(TXD1)*%
%TO.N,/PD3*%
```

and

```
%TO.P,IC1,9,PD3(TXD1)*%
%TO.N,/PD3*%
```

generate the problem.
Changing PD3 to another value fix the issue. Weird bug.

Will see it more in depth later.

netquik, 2022-03-10:
Now I think that "D3" in a net attribute goes somewhere in regex of the parser to looks like a command or such.
Difficult to debug from my side.

netquik, 2022-03-11:
**So far what i found**:
1. Tracespace could  recognize netlist attribute (by regex) as operation
2. Specifically performs a flash operation, creating a flash object at (X,Y) (previous selected object coords) when "D3" match true (`D0*([123])`) 
3. The issue should be present also for D01 or D02 (but probably not performed or not causing visible modification)
4. **The issue can be wider in scope** (with netlist name attributes matching other operations and commands)
5. The issue seems to be linked also to block identification (it rises only after first object selection)
6. The bug is in the parser so **it shows up also in svg generation** (!!!)
7. The issue is not easily easily recognizable visually. In my case I had a IC footprint with 2 pairs of pads array 0° and 90° (like most of IC footprints). So tracespace performs a flash of the last pad of the set (90° or 270°) and duplicates it with a 0° pad in the same coords. Note: if the flash happens on 0° element it is duplicated in same position and angle so it is not visible.

**What i think**:
1. I strongly suggest to **use tracespace only with gerber without netlist attributes**
2. For a fix, it could be done in multiple ways (removing netlist declarations before parsing, identifying them and ignoring, changing block parsing or changing operation regex
3. Speaking about regex, in this case i'm not sure why `D0*([123])` is used (it matches true also if zero is not present), `D0+([123])` would fix my problem but the fact is that we can always have a net name like "D03" (that would rise the issue again).

i will try to create a simpler gerber showing the issue

netquik, 2022-03-11:
Now see this simple gerber
`%TF.GenerationSoftware,KiCad,Pcbnew,(6.0.2)*%
%TF.SameCoordinates,PX420bf58PY2fe28b8*%
%TF.FileFunction,Copper,L1,Top*%
%TF.FilePolarity,Positive*%
%FSLAX46Y46*%
%MOMM*%
%LPD*%
G01*
G04 APERTURE LIST*
%TA.AperFunction,SMDPad,CuDef*%
%ADD10R,0.543500X4.796500*%
%TD*%
%TA.AperFunction,SMDPad,CuDef*%
%ADD11R,9.543000X2.187000*%
%TD*%
G04 APERTURE END LIST*
D10*
%TO.P,BUG1,1,BUG*%
%TO.N,unconnected-(BUG1-Pad1)*%
X50000Y7600000D03*
%TD*%
D11*
%TO.P,BUG2,1,BUG*%
%TO.N,/PD3*%
X25000Y0D03*
%TD*%
M02*
`
It has 2 pads. The first not connected to and the second connected to a PD3 name net.

[image]

Tracespace does flash the second net pad (reading a D03 op) to first pad coords (crossing it).

[image]

netquik, 2022-03-11:
Maybe this change can avoid the problem till further evaluation
A new regex for object attribute and a return on this reg test (as for comments)

_parse-gerber.js
https://github.com/tracespace/tracespace/blob/4dd28dfa5a99512460469ce61430e4803ec434cf/packages/gerber-parser/lib/_parse-gerber.js#L17
```
// object Attributes (TO)
// can be more specific and optimized but should work anyway
var RE_TO = /^%TO[^%*]*/

// tool changes
var RE_TOOL = /^(?:G54)?D0*([1-9]\d+)/
```

_parse-gerber.js
https://github.com/tracespace/tracespace/blob/4dd28dfa5a99512460469ce61430e4803ec434cf/packages/gerber-parser/lib/_parse-gerber.js#L120

```
var parse = function(parser, block) {
  // if comment or object attribute return
  if (RE_COMMENT.test(block) || RE_TO.test(block)) {
    return
  }
```
  
  
  
[image]

netquik, 2022-03-11:
Finally i think some regex should be fixed starting from OP
var RE_OP = `/D0*([123])$/` (should be specific `/D0([123])$/`)
var RE_COMMENT = `/^G0*4/` (should be specific `/^G04/`)
and so var RE_COORD and maybe other

ex: D1 D2 or D3 (or D0001 etc) should not be valid operation codes. D01 D02 and D03 are strictly used

as long as i understand from
https://www.ucamco.com/files/downloads/file_en/416/the-gerber-layer-format-specification-revision-2021-02_en.pdf?#page=18

commit 844d44b9 (2022-03-12): fix(gerber-parser): ignore object attribute %TO blocks (#376)

Fixes #375
closed 2022-03-12 by mcous