tracespace/tracespace #380: "gap fill limit" equivalent in tracespace-cli

ebastler, 2022-04-08:
Hello, since KiCad 6.0.x seemed to start generating gerbers with tiny gaps at some point (projects I exported in earlier kicad versions (6.0.0) draw fine, the exact same projects re-plotted in the current stable (6.0.4) do not) and I am relying on tracespace-cli to render PCB previews, I am trying to achieve the same behavior as the "gap fill limit" feature from the web client in tracespace-cli.

I can set the "Gap fill limit" to 0.01 in tracespace web GUI and it draws the gerbers fine, but I am struggling to achieve the same behavior in the tracespace CLI client. I tried to read into the docs/apis, but am struggling to find a solution.

Full command which I use to call tracespace-cli, maybe it helps:
`tracespace -L -b.color.sm="rgba($colorRGBSM,$colorAlphaSM)" -b.color.ss="rgb($colorRGBSS)" -g.plotAsOutline="0.05" *`
The color variables are expanded by the shell script, the command is called in a folder with gerber files.

What I have tried to set  `-g.plotAsOutline` to:
* "false" - same as no config option. PCB draws, but with incorrect outline due to gaps.
[image]
* "true" - same as no config option
* "0.00011" - supposed to be the default option. Wrong outline + messed up stackup (despite the output still telling me it recognizes all layers correctly)
[image]
* "0.05" - outline is now detected correctly, but all other layers are still messed up.
[image]

I am lost at this point and do not know how to get my script working again. Is plotAsOutline even the correct parameter to achieve my goals? Are my issues caused by me applying the gap correction to all layers, instead of just the outline? If so, is there a solution to only specify one layer for which I want this to apply? Since the web client has a working gap fill limit feature, I suppose that the tool is capable of doing so, I am just missing something.

mcous, 2022-04-08:
Sorry, the CLI options are a bit... byzantine. If possible for your use case, [a config file](https://github.com/tracespace/tracespace/tree/main/packages/cli#options) may be a better option here.

That being said, I believe you're right in your guess that you're accidentally applying the `plotAsOutline` parameter to all Gerber files instead of just the outline file. I think what you're looking for is the `--layer` option rather than the `--gerber` option.

```shell
tracespace ... -l.${outline_filename}.plotAsOutline=0.05
```

As a config file, your full config could be specified as:

```js
// tracespace.config.js
module.exports = {
  noLayer: true,
  board: {
    color: {
      sm: "...",
      ss: "...",
    },
  },
  layer: {
    "outline_filename.gbr": 0.05,
  },
}
```

(Please excuse any typos above, I'm operating from memory and don't have files handy to test on)

ebastler, 2022-04-09:
Thank you very much for your quick help! 
`-l.${outlineFilename}.options.plotAsOutline=0.05` did the trick. Took me a bit to figure out I needed to add the .options part.

I prefer to use parameters only, since I wrote a small bash tool that serves as a wrapper making tracespace/cli a bit easier to use, and I call it as a script in random folders containing gerber files - not needing configs + paths to them makes it a lot more portable as it is only a single file you can symlink into any user /bin folder.

Guess I could create the config file with the script in the gerber folder, but at that point I'd just be making my life unnecessarily difficult.

The CLI options are indeed a bit hard to grasp, but it's still an amazing tool - thanks a lot for the project. Being able to quickly run a quite realistic preview drawing of gerbers is great help.

closed 2022-04-09 by ebastler