tracespace/tracespace #381: Provide ESM build

nosovk, 2022-04-12:
While trying to use Vite bundler in SvelteKit I found that import that defendant to ESM broke build
```typescript
import gerberParser from 'gerber-parser';
import gerberToSvg from "gerber-to-svg";
```
I can provide reproduction repo, but in common there is just no ESM build now. Providing it will allow library usage in Vite, and within dynamic import in browser.

mcous, 2022-04-12:
I do not intend to add ESM builds for the v4 release line, since v4 is written in ES5. I am not familiar with SvelteKit, but I would expect you to be able to configure your bundler to treat certain dependencies as CommonJS.

ESM builds are already enabled for v5, but v5 is a work in progress and I do not have a release timeline. I've published a pre-release version of the parser has been released, if you want to try it out: https://www.npmjs.com/package/@tracespace/parser

nosovk, 2022-05-03:
then will just wait for v5, thanks

kasbah, 2022-06-07:
I've managed to use Tracespace packages with the Vite bundler on a project just now. Not having an ESM build hasn't been a problem for me but there is some incompatibility between Vite and the `readable-stream` package. I've resolved it by adding a this to package.json (I'm using yarn, not sure if the npm cli supports this):

```json
  "resolutions": {
    "readable-stream": "npm:vite-compatible-readable-stream@3.6.1"
  },
```

mcous, 2022-06-29:
I, too, have successfully bundled the existing CJS packages with vite. Going to close this issue since it's possible to bundle (potentially with workaround) the v4 packages and v5 packages are ESM native

closed 2022-06-29 by mcous