tracespace/tracespace #389: Incorrectly displaying triangle shape top and bottom SVG

uugear, 2022-07-08:
After uploading a ZIP file that contains Gerbers generated by Altium Designer to [https://tracespace.io/view/](https://tracespace.io/view/), I suprisingly found the top and bottom SVG are displayed as triangle shape, which is incorrect.

Here is what I saw for the top SVG:
[image]

And here is what I saw for the bottom SVG:
[image]

As for the "layers" tab, it displays quite well.

Could you give some hints about what could cause this kind of issue? Thank you in advance. 

Here is [the ZIP file I uploaded](https://www.uugear.com/tmp/U3.zip).

mcous, 2022-07-08:
Hi @uugear, [this problem](https://github.com/tracespace/tracespace/blob/v4.2.8/apps/view/HELP.md#no-image-or-incorrect-board-shape) is caused by gaps in this board's outline layer.

The outline layer contains the following four line segments, in this order:

| start | end |
| --- | --- |
| `[0, -0.0001]` | `[0, 48.7499]` |
| `[43.14, -0.0001]` | `[43.14, 48.7499]` |
| `[0, 48.7499]` | `[43.13, 48.7499]` |
| `[0, 0]` | `[43.14, 0] |

Note the mismatches between `43.14` <> `43.13` and `-0.0001` <> `0`. The viewer is a little too strict by default when it comes to filling gaps in outline layers. If this is a board of your own design, I would recommend fixing the outline layer so that the lines share their endpoints exactly.

If you can't export a fixed outline, you can also use tracespace's "gap fill limit" setting: increase the number until the board renders correctly.

closed 2022-07-08 by mcous

uugear, 2022-07-11:
Thank you @mcous very much for the help!