tracespace/tracespace #399: Outline parsing problem

demmelproducts, 2022-11-10:
Hi,

the outline of my board is not correctly parsed. 

[image]

This seems to happen because of one of the a circle segments in the outline of the board as it works when I replace this specific circle segment with a manually drawn polygon.

[image]

Please note that in both cases the outline *is* a closed line (I'm aware of the problem whn there are gaps).

The ZIP file contains the two Gerber files
[Gerbers.zip](https://github.com/tracespace/tracespace/files/9979115/Gerbers.zip)

---

Something else: I really would like to have the possibility to show vias displayed as covered (possibly by selecting the max. diameter)

Thank you for your excellent work!
Herbert

mcous, 2022-11-10:
Thanks for the files! It definitely looks like there's at least one actual (tiny) gap in the outline.

[image]

I can get it to render properly by changing the gap fill setting:

[image]

Digging into this a little further, while Gerber has the ability to draw arcs, these Gerber files **do not draw any arcs**! It's all line segments, which points to potentially a problem with your CAD software's output. What software did you use to generate these Gerber files?

Regardless, I appreciate these files as reference, because I have the sneaking suspicion that the outline gap filling logic either isn't correct, is too strict, or both, and I'm looking to improve it. So thanks again!

demmelproducts, 2022-11-10:
Ooops, you got me ;-)
I missed to set this option (simply because I was not aware of it...)

I've made this file with a old version of Eagle (it was an old design which I updated recently). 
I tested the outline by moving every edge to see if the joing line moves accordingly, so Eagle *does* think it's a a closed line.

---

Currently in my company Altium is used instead (but not by myself as I do not want to to learn it anymore), which has not the possiblity of outputting the traces and drills in the STEP file (but it's capable of doing this in a 3D PDF file which can not be converted to STEP...), so I'm considering to use your output data as a "surface image" for generating much more realistic looking STEP files.

So this leads me to me feature request again: 
I really would like to have the possibility to show vias displayed as covered (possibly by selecting the max. diameter) - what do you think about this feature? It would enable us to have a more realistic preview of the PCB.

Anyway, thank you for your help and ultra-fast reply !!!

Best regards
Herbert

mcous, 2022-11-11:
> I'm considering to use your output data as a "surface image" for generating much more realistic looking STEP files.

Oh, very cool!

> I really would like to have the possibility to show vias displayed as covered (possibly by selecting the max. diameter) - what do you think about this feature? It would enable us to have a more realistic preview of the PCB.

This is a really interesting request! Up until this point, we've been applying the _entire_ drill file image as sort of a "removal" on the already rendered board (i.e. copper + solder mask + silk screen). So I think your request definitely exposes an inaccuracy in the render.

In reality (and according to a few sets of Gerber files I just checked), the solder mask layer is generated as explicitly covering the drill hits. Accordingly, we should be rendering the stack-up in a slightly different order. I've renamed this ticket accordingly! This change shouldn't be too hard to make, though I'll be trying it out in the `v5` branch before deciding whether I want to port it back to the (current) `v4` version