tracespace/tracespace #400: Outline SVG is coming blank for a particular Gerber file and that is distorting the top and bottom view

koushikmondal-digikey, 2022-11-21:
One of the user reported that, one of the Gerber file (Altium) they uploaded is not working as expected. The top and bottom view of the Gerber Preview is distorted due to the outline file. When the outline file is removed from the Gerber, the preview working properly.

After debugging though it, we found out that the outline file image from **gerber-to-svg** library is coming as blank (tried rendering the svg output to html img tag). The  This same file is working on the website mentioned below, so its not the issue with the file itself.

Attached is the Gerber outline file and screenshots.

Works fine on this website:
https://www.pcbway.com/project/OnlineGerberViewer.html

Gerber File (contains only the outline file):
[gerber.zip](https://github.com/tracespace/tracespace/files/10057110/gerber.zip)
[image]

kasbah, 2022-11-21:
There are gaps in the outline. You can bridge the gap using the `outlineGapFill` setting. PCBWay's defaults are likely higher than the Tracespace defaults. With 0.1mm it works on tracespace.io:

[image]

koushikmondal-digikey, 2022-11-23:
@kasbah Thanks for the quick response, it worked. Truly appreciate the help!

closed 2022-11-23 by kasbah