tracespace/tracespace #401: Does whats-that-gerber parses and identifies whether slots or cutouts exist in a drill file??

VineethDigiKey, 2022-12-06:
Hi Team,

We want to verify if the drill file contains **_Slots or internal cutout designs_**, would like to know if whats-that-gerber parses the files and provides those details, If yes request you to kindly provide the details how we can read this details using the library. If not please provide some suggestion how we can implement this.

Please let me know if you need more details.

mcous, 2022-12-06:
Hi @VineethDigiKey! As stated in the [whats-that-gerber](https://github.com/tracespace/tracespace/tree/main/packages/whats-that-gerber), it's purpose is to:

> Identify Gerber and drill files by filename

It does not parse files. If you would like to parse files, you should be looking at [gerber-parser](https://github.com/tracespace/tracespace/tree/main/packages/gerber-parser) or [@tracespace/parser](https://github.com/tracespace/tracespace/tree/v5/packages/parser) (prerelease).

Depending on your needs, reading / parsing the drill file may be enough! You should be able to examine the parsed output of a drill file and determine if it uses any `G85` commands (slots) or any `G1/2/3` (interpolate) commands.

However, in my experience _internal cutouts_ (as opposed to simple slots) are usually specified on the [outline layer](https://docs.oshpark.com/submitting-orders/board-outline/#board-cutouts) rather than the drill layer. Extracting information from outline layers is a harder problem that I've been spending a lot of time on recently. I don't have any easy answers here! But in general a solution here might look like:

1. Parse the outline layer
2. Plot the outline layer
3. Examine the plotted outline and try to identify an exterior loop (board shape) and any interior loops (board cutouts)

This analysis will be complicated by the following facts:

- Outline layers are ofter drawn and/or output with paths that are not fully closed in the Gerber, meaning you need to identify and fill small gaps
- Some Gerber output duplicates things like drill holes in both the drill files and the outline files
- Some Gerber output uses linear paths to approximate circles in the outline layer rather than arcs, making it difficult to differentiate between drill holes and internal cutouts in the outline layer

Depending on your accuracy needs, these problems may be insignificant or insurmountable, or maybe somewhere in between!

mcous, 2022-12-08:
It is quite difficult for me to diagnose any particular issue unless you also post the code you wrote to use `gerber-parser`. If you post that code, I may be able to help. 

I would also encourage you read the [API documentation](https://github.com/tracespace/tracespace/blob/main/packages/gerber-parser/API.md) for the package if you haven't already

VineethDigiKey, 2022-12-09:
Hi @mcous, sorry for that, here is the code what we used
this._rawLayers = this._files.map((f) => ({
      filename: path.basename(f.filePath),
      gerber: fs.createReadStream(f.filePath).pipe(parser),
    }));

we get below error at line#3 i.e. when createReadStream is triggered
[image]
we also tried providing only 1 file i.e. the drill file, we get same issue.

mcous, 2022-12-09:
Unfortunately that snippet doesn't give me much to go on. Are you able to either share any of the files you're trying to parse, or create a more minimal, complete reproduction?

For example:

```js
// usage: node index.js [gerber_file]
const fs = require('node:fs')
const assert = require('node:assert')
const createParser = require('gerber-parser')

const inputFile = process.argv[2]
assert(typeof inputFile === 'string', 'Argument must be a path to a Gerber file')

const inputStream = fs.createReadStream(inputFile)
const parser = createParser()

inputStream
  .pipe(parser)
  .on('data', console.log)
  .on('warning', console.warn)
  .on('error', console.error)
```

If I run this against the following drill file:

```
M48
INCH,TZ
T1C0.5
%
G90
T1
X0Y0
M30
```

...then I get the following output...

```
❯ node index.js ./drill.drl
{ type: 'set', line: 2, prop: 'units', value: 'in' }
{
  type: 'tool',
  line: 3,
  code: '1',
  tool: { shape: 'circle', params: [ 0.5 ], hole: [] }
}
{ type: 'set', line: 5, prop: 'nota', value: 'A' }
{ type: 'set', line: 6, prop: 'tool', value: '1' }
{ type: 'op', line: 7, op: 'flash', coord: { x: 0, y: 0 } }
```

VineethDigiKey, 2022-12-12:
Hi @mcous, this was helpful.
 
I tried with your code, and I am also able to get similar output for my drill and outline files. Below output is form outline file.

Our requirement is to validate if drill file contains Slots or internal cutout designs, our understanding is Slots can be identified in the drill file, the file is using traces.  Internal Cutouts are areas that are cutout in the outline file or fab drawing.

I would like to understand on the basic of this output how to figure out whether this drill/outline file contains Slots or internal cutouts, I see that we have holes but not sure if that is same as Slots.

I am not able to attach the file(drill or outline). Please let me know how I can share

```
{ type: 'set', line: 32, prop: 'mode', value: 'ccw' }
{
  type: 'op',
  line: 33,
  op: 'int',
  coord: { x: 97.79, y: -114.3, i: 3.81, j: 0 }
}
{ type: 'set', line: 34, prop: 'mode', value: 'i' }
{ type: 'op', line: 35, op: 'move', coord: { x: 195.58, y: -114.3 } }
{ type: 'set', line: 36, prop: 'arc', value: 'm' }
{ type: 'set', line: 37, prop: 'mode', value: 'ccw' }
{
  type: 'op',
  line: 38,
  op: 'int',
  coord: { x: 199.39, y: -110.49, i: 0, j: 3.81 }
}
{ type: 'set', line: 39, prop: 'mode', value: 'i' }
{ type: 'op', line: 40, op: 'move', coord: { x: 199.39, y: -39.37 } }
{ type: 'set', line: 41, prop: 'arc', value: 'm' }
{ type: 'set', line: 42, prop: 'mode', value: 'ccw' }
{
  type: 'op',
  line: 43,
  op: 'int',
  coord: { x: 195.58, y: -35.56, i: -3.81, j: 0 }
}

```

mcous, 2022-12-12:
You can add Gerber and drill files to GitHub issues by putting them inside a `.zip` archive first, if you are comfortable sharing them.

It will be difficult to determine if drill/outline layers have cutouts just from looking at the parsed output. It will be easier to look at the plotter output. For that, you can use `gerber-plotter`.

```js
const fs = require('node:fs')
const assert = require('node:assert')

const createParser = require('gerber-parser')
const createPlotter = require('gerber-plotter')

const inputFile = process.argv[2]
assert(typeof inputFile === 'string', 'Argument must be a path to a Gerber file')

const inputStream = fs.createReadStream(inputFile)
const parser = createParser()
const plotter = createPlotter()

inputStream
  .pipe(parser)
  .pipe(plotter)
  .on('data', data => console.log('%o', data))
  .on('warning', console.warn)
  .on('error', console.error)
```

If I take [an outline file](https://github.com/tracespace/tracespace/blob/main/packages/fixtures/boards/clockblock/clockblock-Edge_Cuts.gbr) and pass that through the parser and plotter, I will get output that looks like this:

```js
❯ node index.js ./outline.gbr
{
  type: 'stroke',
  width: 0.005,
  path: [
    {
      type: 'line',
      start: [ 4.1, 3.6, [length]: 2 ],
      end: [ 4.1, 3.7, [length]: 2 ]
    },
    {
      type: 'line',
      start: [ 4, 3.7, [length]: 2 ],
      end: [ 3.2, 3.7, [length]: 2 ]
    },
    {
      type: 'line',
      start: [ 4, 3.6, [length]: 2 ],
      end: [ 4, 3.7, [length]: 2 ]
    },
    {
      type: 'line',
      start: [ 3.2, 3.6, [length]: 2 ],
      end: [ 4, 3.6, [length]: 2 ]
    },
    {
      type: 'line',
      start: [ 3.2, 3.7, [length]: 2 ],
      end: [ 3.2, 3.6, [length]: 2 ]
    },
    {
      type: 'line',
      start: [ 3.1, 3.7, [length]: 2 ],
      end: [ 3.1, 3.6, [length]: 2 ]
    },
    {
      type: 'line',
      start: [ 1.55, 3.7, [length]: 2 ],
      end: [ 3.1, 3.7, [length]: 2 ]
    },
    {
[... 189 lines cut ...]
```js
{
  type: 'shape',
  tool: '1',
  shape: [ { type: 'circle', cx: 0, cy: 0, r: 0.25 }, [length]: 1 ]
}
{ type: 'pad', tool: '1', x: 0, y: 0 }
{
  type: 'stroke',
  width: 0.5,
  path: [
    {
      type: 'line',
      start: [ 0, 0, [length]: 2 ],
      end: [ 0.5, 0, [length]: 2 ]
    },
    [length]: 1
  ]
}
{
  type: 'size',
  box: [ -0.25, -0.25, 0.75, 0.25, [length]: 4 ],
  units: 'in'
}
```

VineethDigiKey, 2022-12-19:
Hi @mcous, 
Thanks for the response, 
please find the 2 drill and 1 outline file [here](https://gist.github.com/VineethDigiKey/d1eb5358744a3218dc578ef49b1dad78), please click on the download zip button to get the files.

Its not very clear how I can figure out if the file contains cutouts or slot, actually I am planning to create two methods which would give me a Boolean value saying file contains cutouts and another method for slots.

As you mentioned in your comments using strokes and pad we can do that, could you please let me know exactly how I can get if file contains cutouts and slots using strokes and pad.

Thanks in advance!

mcous, 2022-12-20:
Hi @VineethDigiKey, I agree, you'll need (at least) two different procedures:

1. Determine if a drill file has slots
2. Determine if an outline file has cutouts

To date, all that tracespace does is attempt to **render** a board shape from an outline layer. Thanks to how SVG works, this is a _much_ simpler problem that trying to truly understand the geometry of a drill layer or an outline layer and do something like count slots.

This is the most detailed set of ideas about how to accomplish slot counting that I am going to be able to give you. You're going to have to figure out the rest on your own! I'm going to close this issue as "answered to the best of my ability," but please feel free to continue asking questions in this thread, and I'll do my best to keep answering them. 

So, the first way I would try to approach this problem would be:

### 0. Figure out if a file is a drill file or an outline file

- Use the output from the `whats-that-gerber` and `gerber-parser` for this determination
- `whats-that-gerber` will guess the layer type from the file name
- `gerber-parser` will tell you if the file parsed as a Gerber file or a drill file

For example, here's how the [tracespace CLI identifies layers](https://github.com/tracespace/tracespace/blob/49c9a22addccc70c1d1a830a9225328a91942080/packages/cli/lib/cli.js#L60-L64)

### 1. Determine if a drill file has a slot

1. Send the file through `gerber-parser` and `gerber-plotter`
2. Every time you see a graphic object of `type: 'stroke'`, that's at least one slot
3. Examine the start and end points of the path segments in the `path` array to determine if it is one continuous slot or several slots
    - The `start` and `end` objects are tuples (arrays) of the format:
    - `[x: number, y: number]` for line segments and `[x: number, y: number, arcAngle: number]` for arcs
    - Compare a segment's endpoints to the endpoints of other segments to determine if it is part of a larger, continuous path

### 2. Determine in an outline has cutouts

1. Send the file through `gerber-parser` and `gerber-plotter`
2. Find all the `type: 'stroke'` graphic objects
3. Examine the `start` and `end` points of the path segments to identify complete "loops"
    - A loop is a set of segments that create a path that starts from and ends at the same point
4. Given a set of "loops", identify the largest loop; this is likely the board outline
5. All other loops, provided they are contained within the outline loop, are likely cutouts

### Challenges you will likely face

`gerber-plotter` outputs paths exactly as they appear in the Gerber file, which means they are not guaranteed to be in any sort of order or drawn in a consistent direction. The `path` segments you can get from a Gerber file will be determined by how the author chose to draw their outline layer and how their CAD software chose to output it.

I have encountered the following issues when trying to use the outline layer to render a board shape:
 
- two geometrically contiguous segments may not appear next to one another in the `path` list, so you must compare endpoints yourself
- path segments may share a `start` point or share an `end` point rather than the `end` of one segment always lining up with the `start` of the next, so you need to be prepared to "reverse" paths
- the user may not have drawn their lines correctly, or the CAD software may have output them poorly and/or at low resolution, leading to issues like:
    - Small gaps in the outline, which you will need to "fill in" somehow depending on your own heuristics for if two end points are "close"
    - Duplicate line segments in the outline

Good luck, and if you make any progress, please send it my way!

closed 2022-12-20 by mcous