tracespace/tracespace #403: outline layer is not rendered correctly.

shawon1220, 2023-01-12:
Hi, this file can't rendering outline layer correctly.

outline file:
[gko.zip](https://github.com/tracespace/tracespace/files/10399759/gko.zip)

tracespace rendering result:
[image]

correct rendering result:
[image]

mcous, 2023-01-13:
Thanks for the report! There appears to be a bug in the v4 outline plotter. I suspect that the problem is here:

https://github.com/tracespace/tracespace/blob/49c9a22addccc70c1d1a830a9225328a91942080/packages/gerber-plotter/lib/path-graph.js#L44-L52

It looks like the condition is missing an `&& target.type === 'line'`

kasbah, 2023-02-03:
Should we keep this open since it's still a bug in v4?

gecheng971229, 2025-04-11:
Hello, how does the v4 version solve this outline layer clipping bug?