tracespace/tracespace #404: drill layer is not rendered correctly disturbed by origin

shawon1220, 2023-01-13:
Hi, I think **G93X0Y0** just is a origin，not be coordinates

drl file:
[NC_drill.zip](https://github.com/tracespace/tracespace/files/10407853/NC_drill.zip)

M48
;FILE_FORMAT=2:4
INCH,LZ
FMAT,2
ICI,OFF
;TYPE=PLATED
T1C.01
T2C.02756
T3C.04016
%
M72
G90
G93X0Y0
T1C.01F095S3
X019203Y016036
X018136Y01932
X018014Y019908
X017096Y022018
X017687Y022117
X018176Y023499
...

mcous, 2023-01-13:
Thanks for the report! The offending regex is here:

https://github.com/tracespace/tracespace/blob/49c9a22addccc70c1d1a830a9225328a91942080/packages/gerber-parser/lib/_parse-drill.js#L21

I don't recall why the regex does not start with `^`, but it probably should, and I think that would fix the issue

mcous, 2023-01-15:
Duplicate of #353

closed 2023-01-15 by mcous