tracespace/tracespace #411: Error: File type not recognized in v5

shawon1220, 2023-02-03:
Hi, Excited to use the V5 version，But the file is uploaded by the user. There is no way to know whether it is a reasonable gerber file or a drill file. Can these files be ignored instead of throwing an exception?

kasbah, 2023-02-03:
I've not got the details but can't you wrap it in a `try`/ `catch`?

mcous, 2023-02-03:
Thanks for the feedback @shawon1220! Since v5 is still an `alpha`, APIs are going to change - some drastically - prior to final release. If you have any feedback like this as you use the alpha packages, I'd love if you could add your feedback in the [alpha feedback discussion thread](https://github.com/tracespace/tracespace/discussions/412)

closed 2023-02-03 by mcous