tracespace/tracespace #414: About Allegro CAD  drill's places

shawon1220, 2023-02-06:
Hi, About Allegro CAD drill's places，Is it possible to be [2,5] when the unit is inches, and [3,5] when the unit is metric?
eg:
[A1000evb_daughter_cards_v1-1-4.zip](https://github.com/tracespace/tracespace/files/10660330/A1000evb_daughter_cards_v1-1-4.zip)

[image]
[image]

shawon1220, 2023-02-06:
BBQ,This case has nc_param.txt and the places is [5,6],
[nc_param.txt](https://github.com/tracespace/tracespace/files/10660493/nc_param.txt)
[9914ce01e-1-8.zip](https://github.com/tracespace/tracespace/files/10660506/9914ce01e-1-8.zip)

mcous, 2023-02-06:
Allegro's drill file output is quite bad, and units / coordinate format default decisions for v5 are still being made. I appreciate the feedback!

Fabs like OSHPark recommend that you select "Enhanced Excellon format". I'm not very familiar with Allegro, so I don't know what the difference is, but it may help.

Tracespace does not support `nc_param.txt` at the moment. That issue is being tracked in #392

closed 2024-03-09 by shawon1220