tracespace/tracespace #420: Maxiumum call stack exceeded when reading some Gerber files

veggiesaurus, 2023-09-08:
I've been using tracepsace for reading Gerber files generated by kicad v7, and ran into an issue with this file in particular. I've attached the offending Gerber file as well as the source kicad PCB.

The following snippet results in an error:

```typescript
import { read } from "@tracespace/core";

async function main() {
  const readPaths = ["./boatcontrol-F_Silkscreen.gbr"];

  const readResult = await read(readPaths);
  console.log(`Found ${readResult.layers.length} layers.`);
}

main()
  .then(() => console.log("Done"))
  .catch(console.error);
```

[boatcontrol-F_Silkscreen.gbr.zip](https://github.com/tracespace/tracespace/files/12563680/boatcontrol-F_Silkscreen.gbr.zip)
[boatcontrol.kicad_pcb.zip](https://github.com/tracespace/tracespace/files/12563683/boatcontrol.kicad_pcb.zip)

veggiesaurus, 2023-09-08:
Just noticed that this has been fixed by https://github.com/tracespace/tracespace/commit/5496ac9f9ea6d66672864304dd26295b3e1447ef

Any chance of bumping the package version after this bugfix?

Gianarci, 2024-02-20:
> Just noticed that this has been fixed by [5496ac9](https://github.com/tracespace/tracespace/commit/5496ac9f9ea6d66672864304dd26295b3e1447ef)
> 
> Any chance of bumping the package version after this bugfix?

Nice question. Is this great library a dead project? It's really a huge shame, because AFAIK is the best gerber-to-svg rendering library for Javascript. Isn't it?