tracespace/tracespace #431: Example of using in a browser

r-bt, 2025-01-20:
Hi,

First thanks for the great librarty! The SVG renders look great.

I'm trying to use v4 in a browser environment. Specifically I have something like this:

```javascript
const gerberFiles = whatsThatGerber(files.map((file) => file.name));

const layers = files.map(async (file) => {
    const gerber = gerberFiles[file.name];

    const { side, type } = gerber;
    const options = {
        plotAsOutline: type === whatsThatGerber.TYPE_OUTLINE,
    };

    const fileData = await file.text();

    gerberToSvg(fileData, options);
});
```

where files is `File[]`. However I am getting the following error:

```
gerber-to-svg.js?v=c72f4716:3412 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'call')
    at PlotterToSvg.Readable (gerber-to-svg.js?v=c72f4716:3412:14)
    at PlotterToSvg.Duplex (gerber-to-svg.js?v=c72f4716:2968:16)
    at PlotterToSvg.Transform (gerber-to-svg.js?v=c72f4716:4082:14)
    at new PlotterToSvg (gerber-to-svg.js?v=c72f4716:7330:17)
    at gerberConverterFactory (gerber-to-svg.js?v=c72f4716:7578:23)
    at GerberVisualisation.tsx:48:9
```

It looks like the library is expecting fileData to be a `ReadStream` as opposed to a string. I was just wondering what's the correct way of using this library in a browser is?

mcous, 2025-01-20:
The `v4` release uses the [`filereader-stream`](https://www.npmjs.com/package/filereader-stream) package to turn `FileReader` values into streams. You could also try using the [higher-level `pcb-stackup`](https://github.com/tracespace/tracespace/blob/main/packages/pcb-stackup/API.md) package which accepts strings.

For reference, the `v4` version of [tracespace view](https://tracespace.io/view) runs its conversions in a WebWorker - [source code](https://github.com/tracespace/tracespace/blob/49c9a22addccc70c1d1a830a9225328a91942080/apps/view/src/render/worker.ts)

r-bt, 2025-01-20:
Awesome thanks for the info!

closed 2025-01-20 by r-bt